ucfai/knightros-gambit #5: Make project webpage

nashirj, 2021-11-08:
Something like this: https://stanfordstudentrobotics.org/pupper

Could have UCF AI link to our project page on this page: https://ucfai.org/groups/projects/
Could have UCF IEEE link to our project page on this page: https://ieee.cecs.ucf.edu/projects/

nashirj, 2021-11-08:
Things we could include on the website right now:
1) CAD of the board design
2) Electrical schematic
3) Link to this github repo
4) Diagrams showing code structure?
5) Information about the project
6) Member info

nashirj, 2021-11-15:
Reference http://www.raspberryturk.com/, would be cool to eventually have something like that

RuthShryock, 2021-11-22:
I used Figma to make a prototype and here is the link: https://www.figma.com/file/MXUHeaBronQ7sH9lyn8tcR/KGWebsite?node-id=0%3A1

nashirj, 2021-11-22:
We will be making the project webpage in [this repository](https://github.com/ucfai/knightr0s-gambit-website).

nashirj, 2021-11-22:
I lied, we'll be making it in this repository in the /docs folder and using Jekyll (like Raspberry Turk)

closed 2021-12-07 by nashirj