ucfai/knightros-gambit #39: Discuss how player promotions works

nashirj, 2021-11-19:
We need a way to indicate to the player which promotion piece they are allowed to use.

Maybe the promotion pieces can be in a special location on the board, and the graveyard is separated from the promotion pieces area. Then, when pieces are taken from the promotion pieces area, they are backfilled from the graveyard (if there are enough pieces in the graveyard available).

LSaber36, 2021-11-19:
Sounds good. I believe we also plan on using a monitor, so we can indicate that there too.

nashirj, 2021-12-05:
See [project overview graveyard section](https://docs.google.com/document/d/1CPY9yEVWDVO99mbckBnCJi-gKKobkBnB4XnjrLc4HFs/edit#heading=h.gwic8x2awnik) for how this will work

closed 2021-12-05 by nashirj