ucfai/knightros-gambit #73: Set up virtual end stop

nashirj, 2021-12-23:
Need to account for only having one end stop in code by virtually tracking the side of the axis that doesn't have an end stop.

LSaber36, 2022-01-18:
Can only be completed when issue #88 is completed

LSaber36, 2022-02-03:
Update: Need to add new endstops to the code. This includes new pins, pin initializations, and checks in the for loops.

LSaber36, 2022-02-09:
We also need to update the homeAxis() function to account for the new conditions and set currentX and currentY accordingly

LSaber36, 2022-02-10:
- We need to add new endstops to the code. This includes new pins, pin initializations, and checks in the for loops
- We also need to update the homeAxis() function to account for the new conditions and set currentX and currentY accordingly
- We also want to move the tracking out of the loops and compute them directly, after the loop
_**These groups of things should be in their own commits to group the changes**_

nashirj, 2022-02-13:
when you do step 4, please add a comment above the line where you call `home()` explaining that the function aligns axes and sets the currentX and currentY to zero.

closed 2022-02-23 by nashirj