ucfai/knightros-gambit #100: Figure out how to detect corners of board

nashirj, 2022-01-19:
We have three main options:
1) Hard code the image corners into our game loop code based on inspection
2) Make the corners distinct colors and identify them with CV
3) Use some 3rd approach to detect corners (e.g. Harris Corner detector)

My preference is option 2, but we need to discuss with hardware team if we're doing that approach.

ColonelToad, 2022-02-02:
Would using a Harris Corner not be the most accurate way to detect corners? Or is there another reason it's not necessarily the preferred option?

nashirj, 2022-02-02:
[image]
[image]

ColonelToad, 2022-02-13:
So you can see the results of running all the tests, overall Canny has the best accuracy however I was surprised to see Shi-Tomasi spotted pieces off the board as well. 
[image]
[image]
[image]

nashirj, 2022-02-13:
This looks great! Nice progress

ColonelToad, 2022-02-14:
So when using the new board image I'm getting much different results. I'll play around with the values more but Shi-Tomasi is not giving me anything at all. Canny is showing some of the edges on the area of the board where the board isn't under the light, and the gradients from the sobel so the same thing. I actually tried Harris and got some results with it though.
[image]
[image]
[image]

ColonelToad, 2022-02-15:
After resizing the image to 800x800 I definitely saw an improvement with the results. With goodFeaturesToTrack() I used 150 as the parameter and that essentially placed a dot around all the tiles of the board as well as the board edge and the pieces. Granted we can experiment with this value but certainly, 150 is a good maximum for how many features to track. 
[image]
[image]
[image]