ucfai/knightros-gambit #116: Update how piece captures behaves in game loop (for engine/cli)

nashirj, 2022-01-25:
If it's a capture:
1) Send captured piece to the proxy temp spot (need to determine which direction capturing piece is coming from and pick a corner that is out of the way).
2) Send capturing piece to destination square.
3) Send captured piece to graveyard.

Need to update serial messaging scheme to have "~<MOVE_TO_TEMP><START_SQ><LL/LR/UL/UR><MOVE_COUNT>". This issue should be addressed on both the Arduino and Pi sides.

nashirj, 2022-01-26:
Chase pointed out correctly that for step 3 we need to be able to specify in the message to the Arduino from the Pi which corner the captured piece is on. I can think of two main ways to do this.

1) Create a new opcode for steps 1 and 3 to specify that the message should be interpreted as double resolution (where the to-from are given as A-Y, which maps to 0-24) and that would let us specify putting on/retrieving from corners. This requires more logic on the Arduino side to parse the message and make it fit into the current movement scheme (which expects that the Pi will always send moves specifying a row and col that specify the center of the square). This might require writing new movement functions that can start movement from a corner (which requires double resolution).
2) Have all moves sent from the Pi to the Arduino be at this double resolution. This greatly simplifies parsing on the Arduino side and will be easy to implement on the Pi side.

Do you have any thoughts on this @adenm-10 @chasesza @LSaber36 @JohnGierlach ? I prefer option 2.

nashirj, 2022-01-26:
This image shows how the conversion would work and hopefully helps clarify the problem with the current message passing scheme.

[image]

LSaber36, 2022-01-26:
Long story short, I agree with you 100%. The Arduino doesn't need to know about whole spaces anyway really. We need to consider how many spaces to travel to stay on lines for moveAlongEdges(), but that's simple math honestly. Keeping the Arduino movements in terms of the unitSpaces would not only be less costly, but actually reduces complexity from what Aden and I were going to write for the primitive functions. It actually eliminates the need for functions that can do either whole or half piece spaces, since one function can do both if row and col are in terms of the unitSpaces. This also allows us to keep our primitive movement functions for the most part, and just get rid of the multiply by 2 part. From what I understand, the only other change is the raspberry pi multiplying the current row and col vars by two before sending. Apart from some edge cases that may need some mild Arduino computation, this is all around better.

So basically yes, this is a better approach in my opinion. Once we are agree on this decision, we can continue working on our code and get it ready for next week :)

nashirj, 2022-01-26:
Sounds good! Option 2 it is.

closed 2022-02-04 by nashirj