ucfai/knightros-gambit #142: Arduino and Pi should be able to retransmit last message in event of parsing error

nashirj, 2022-02-05:
This will require a new opcode on both sides to request a re-transmission of the last message.

nashirj, 2022-02-13:
I think one way of doing this on the Pi side is to have Arduino send an ERROR status, and set the "Extra" field to a code that indicates we want to retransmit the last message. I can get that set up on the Pi side, lmk if that works for you @chasesza

chasesza, 2022-02-16:
This is a good idea for sure. I think the "Extra" field we should use for this is the "Incomplete_Instruction" error code. 
The Arduino side is already set up if we use that code.

nashirj, 2022-02-22:
After current processing loop finishes (i.e. movement, CV, AI, whatever, send retransmit request before any other messages. We will need to update move count logic; can probably simplify since all we use move count for is an ack. Maybe we alternate between 0 and 1 for all other moves, but if it's a retransmit, the "move number" is "R" for retransmit

nashirj, 2022-02-24:
This issue can be handled as part of #173, which more generally cleans up the opcodes to separate non-chess move type commands into `debug` and `info` type ops

closed 2022-02-24 by nashirj