ucfai/knightros-gambit #151: Set up figshare for storing data and models

nashirj, 2022-02-10:
Use https://figshare.com/ to store our models and data. It's what sklearn uses, and we can set up something similar to how they fetch datasets: https://github.com/scikit-learn/scikit-learn/blob/7e1e6d09bcc2eaeba98f7e737aac2ac782f0e5f1/sklearn/datasets/_california_housing.py#L53

ryPattillo, 2022-02-10:
One of us needs to create an account. We can do that at the meeting tonight.

fshcat, 2022-02-17:
just making sure, do we want the code to automatically write to the figshare or just read from it and we just manually upload new dataset if necessary

nashirj, 2022-02-22:
> just making sure, do we want the code to automatically write to the figshare or just read from it and we just manually upload new dataset if necessary

This is a good question and honestly I'm not sure. I think we should discuss this more at tomorrow's software meeting. I think it depends on if we are iteratively building the dataset or we are just going to build a few different datasets then merge them in a separate step to train on with mcts

nashirj, 2022-02-24:
We just talked about this and there are two options:
1) Build dataset all at once: this would mean we don't need to overwrite, can just download and merge and do mcts training
2) Iteratively go back and forth: this would allow incremental model training, but it adds a lot of complexity with file writing/reading/merging.

**decision**: create dataset all at once; download and train after as necessary

nashirj, 2022-02-24:
tl;dr: We need to add logic in `train.py` for
1) Writing data to figshare after creating it in `create_dataset` (we can do a proof of concept in experimental work)
2) Downloading data from figshare in the event that we want to use existing data (need to update hyperparameters file to use figshare user path and fetch all datasets from user)
3) Merging downloaded data from figshare

This can be done in one PR

ryPattillo, 2022-04-03:
I am working on this right now ^ . Should we have something in params.json that specifies uploading to figshare? Or we always going to upload to figshare after creating the dataset?

ryPattillo, 2022-04-03:
Same with models

ryPattillo, 2022-04-03:
Adding a figshare boolean to params.json

ryPattillo, 2022-09-05:
This has been completed.

closed 2022-09-05 by ryPattillo